Zapit-Optostim/zapit #103: Create a state file that can be reloaded

raacampbell, 2023-02-03:
Imagine Zapit needs to be stopped and restarted for some reason but nothing changed in between. So there is no reason to re-do any calibrations. It would be great if we could re-apply the last state of the software before it shut down the last time. This would speed up development but I can also see it helping users too. Probably there is only a need to bring back the last state. So we can have a single menu button to do just that. Maybe the state is saved in a `.mat` file in the settings directory?

raacampbell, 2024-03-11:
@njmiska A couple of short scripts to implement the loading and saving steps would look like this:

**To store the calibration data**

```
t_form = hZP.transform; % scanner calibration transform%

refPointsSample = hZP.refPointsSample;
refPointsStereotaxic = hZP.refPointsStereotaxic;

% Store these to a mat file
save('backupSettings.mat', t_form, refPointsSample, refPointsSample)
```

**To re-apply previously loaded data**
```
load('backupSettings.mat')

hZP.transform = t_form
hZP.scannersCalibrated = true;

refPointsSample = hZP.refPointsSample;
refPointsStereotaxic = hZP.refPointsStereotaxic;
obj.sampleCalibrated = true;
```

njmiska, 2024-03-11:
I think the line to save the settings needs to be:

save('backupSettings.mat', 't_form', 'refPointsSample', 'refPointsSample')

I was not able to successfully load either the scanner or sample calibration using the second set of code. Sample calibration simply did not show up (and 'light' is still red), with no error. Scanner calibration initially seems to load, but then when I click 'test calibration' it throws an error and the scanner calibration goes back to red.

raacampbell, 2024-03-13:
There was a typo also on the last line. I updated it. 
Can you try testing the scanner calibration by instead doing "cat and mouse"? The above line saves the transform matrix only. When you press "test calibration" you are asking the software it to run over a grid of points used in the calibration. Above we didn't save that, so that will be the cause of that error. However, I _think_ the transform matrix is all you need to move forward. i.e. that will convert a mouse position to a laser location.