Zapit-Optostim/zapit #149: Rampdown might possibly be starting one cycle too

raacampbell, 2024-02-06:
We asked for 800 ms plus 250 ms for Oli's figure and we get the following:
[image]

The data are measured with a photodiode and it's possible the start of the rampdown is being missed due to saturation. Over all it is running for the correct time period but maybe we need to tweak how the ramp down happens. 

This may be a minor bug or may be a measurement error.

raacampbell, 2024-02-06:
I start in simu mode version 1.02 on `dev`. I load `uniAndBilateral_5_conditions.yml` then:
```
[~,~,W]=hZP.sendSamples('cond',2,'stimDurationSeconds',0.8);
```

The results look as expected:
[image]

Possibly @omgauld001 had an older version with a bug? Or possibly the photodiode was saturated? The lack of the last pulses might be due to the fact that the laser cuts off at very low powers.

omgauld001, 2024-02-08:
We've been using version 1.0.2. in the matlab window it states 'Zapit is up to date'

omgauld001, 2024-02-08:
[image]
This was a measurement taken in Aug 2023. The stim is 500 ms and then 250 ms rampdown. The rampdown starts at the right time, but again it seems like the last laser pulse is not delivered.

raacampbell, 2024-02-08:
> We've been using version 1.0.2. in the matlab window it states 'Zapit is up to date'

You probably are pulling from the `dev` branch, since `master` is v1.0.2 (https://github.com/Zapit-Optostim/zapit/releases). Just keep that in mind. 

Looks like you had saturation originally and the failure to see the last pulse is just the laser not being able to produce a super-low power.